florentbr/OWON-VDS1022 #97: Different results between the official application and Python project

yurii-vavrynchuk-embrox, 2024-05-17:
Hello,
I have the following issue:
I setup the official application in that way that I need (5 V/dif), I set the same config for Python project (50V to be the same), but my data   comes with the very large scale. Can you advise me how to set your project to retrieve the exact same results as official application?
Best regards, Yurii

florentbr, 2024-05-17:
Have a look at [vds1022.ipynb](https://github.com/florentbr/OWON-VDS1022/blob/master/api/python/vds1022.ipynb) for examples.
Calling `frame.ch1.y()` should get you the same scale as the official app.

yurii-vavrynchuk-embrox, 2024-05-18:
@florentbr thanks for your answer! I want to make sure that I correctly understand everything.
I have my flash in oscilloscope ruined (I had opened the appropriate issue yet). Does it mean that all values in the official are invalid? Because I can't see any official calibration and a make a assumption that my calibrations in the official calibration are invalid at all.

florentbr, 2024-05-18:
> Does it mean that all values in the official are invalid? 

The official app read the calibration from flash too. If you are not getting the right one with the python API, then the official app won't.
There's a hidden menu in the official app to tune the calibration. I don't remember the exact shortcut. Could be Ctrl+L, Ctrl+U or ATlt+L.
Note that when first launched, the official app creates a backup of the calibration somewhere in the user folder.

yurii-vavrynchuk-embrox, 2024-05-18:
@florentbr thanks, I have received the correct values for now!
I have commented flash reading in your code and read the backed up calibration from the folder. The main problem in my application was that I set probe x10 which performs strange translation of my graph, for now I am using x1 and results are exactly the same as my power supply provides.