florentbr/OWON-VDS1022 #98: Installing on Ubuntu : Unmet dependencies

Craigp23, 2024-06-25:
Needing some help installing Owon-VDS1022 on Ubuntu 22.04. When running sudo bash install-linux.sh I'm seeing...

```
===========================================================
 Build package                                             
===========================================================
Check environement ...
Add program files ...
Add desktop menu ...
Add usb permissions ...
Build debian package ...

Package:
/tmp/owon-vds-tiny-1.1.5-cf19.amd64.deb

===========================================================
 Install package owon-vds-tiny-1.1.5-cf19.amd64.deb
===========================================================
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies.
 bitscope-dso : Depends: libgdk-pixbuf2.0-0 (>= 2.22.0) but it is not installable
                Depends: libpango1.0-0 (>= 1.14.0) but it is not installable
 owon-vds-tiny : Depends: java-runtime but it is not installable
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

I've attempted to install libgdk-pixbuf2.0-0 but said not available, try libgdk-pixbuf-2.0-0

Been going round in circles for a few hours - any tips very welcome. TIA :)

florentbr, 2024-06-25:
Regarding the missing `java-runtime`, it's an old alias. It's probably no longer in the repository.
Try to replace it by `default-jre` or any available alias for a Java runtime in `install-linux.sh`.
The `bitscope-dso` has nothing to do with this project, just ignore it.

Craigp23, 2024-06-26:
Thanks for the tip! I've searched for advice on how to use `default-jre` but still not clear. Is it used as a parameter on the` install-linux.sh` command line ?

florentbr, 2024-06-26:
> Thanks for the tip! I've searched for advice on how to use `default-jre` but still not clear. Is it used as a parameter on the` install-linux.sh` command line ?

It's not a command line, it's text in the `install-linux.sh` file .
Anyway I just checked and the package `java-runtime` is still present with Ubuntu 22.04:
https://pkgs.org/search/?q=java-runtime&on=provides

There's something wrong with your package manager.
You'll have to fix it first or install the packages `default-jre` and `/tmp/owon-vds-tiny-1.1.5-cf19.amd64.deb` manually.

Craigp23, 2024-06-26:
Thanks again! 

I'm stuck unfortunately :/ I tried `sudo apt install default-jre` which returned...

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package default-jre is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'default-jre' has no installation candidate```

Any tips appreciated!

florentbr, 2024-06-26:
Your issue is likely due to a corrupted repository list. Restore it to Ubuntu Main.
https://www.google.com/search?q=ubuntu+reset+repository+list

florentbr, 2024-06-26:
Closing since the issue has nothing to do with this project.

Craigp23, 2024-06-26:
> Your issue is likely due to a corrupted repository list. Restore it to Ubuntu Main.
> https://www.google.com/search?q=ubuntu+reset+repository+list

This resolved the issue on this machine. VDS-1022 now installed. Thanks for your assistance! :)