florentbr/OWON-VDS1022 #100: UI freezes on offset or trigger slide

ikcom, 2024-08-04:
The UI freezes at any attempt to slide the offset or trigger rulers.
Software doesn't respond to window close either, only process kill.

```text
env: Linux, Java 1.8.0_422 amd64
app dir: /opt/owon-vds-tiny
user dir: /home/lab/.owon-vds-tiny
locale: en, en_US, false
[Send @MACHINE_TYPE_ADD 0x56] recv V 0x1
found usb device VDS1022 idVendor:0x5345 idProduct:0x1234
[Send @FPGA_DOWNLOAD_QUERY_ADD 0x0] recv E 0x0
fpga loaded: 0
read flash
[Send @read_flash 0x1] recv 2002 bytes
flash size: 2002
flash crc32: 1435005807
flash header: 0xAA 0x55
flash version: 2
flash oem: false
flash board version: V2.5
flash board serial: VDS1022I1726768
load calibration /home/lab/.owon-vds-tiny/VDS1022I1726768-cals.json
Gain CH0 : [1249, 1098, 969, 798, 672, 543, 976, 847, 719, 548]
Gain CH1 : [1292, 1099, 970, 801, 674, 546, 975, 847, 720, 550]
Step CH0 : [154, 152, 153, 154, 153, 153, 155, 155, 152, 153]
Step CH1 : [153, 153, 152, 153, 151, 153, 153, 153, 153, 153]
Compensation CH0 : [527, 532, 539, 539, 539, 541, 539, 539, 536, 541]
Compensation CH1 : [512, 524, 528, 530, 531, 533, 529, 530, 530, 533]
load fpga /opt/owon-vds-tiny/fwr/VDS1022_FPGAV1_V3.9.bin
[Send @FPGA_DOWNLOAD_ADD 0x24832] recv D 0x2800
[Send FPGA part 1/15 ] recv: S 0
[Send FPGA part 2/15 ] recv: S 1
[Send FPGA part 3/15 ] recv: S 2
[Send FPGA part 4/15 ] recv: S 3
[Send FPGA part 5/15 ] recv: S 4
[Send FPGA part 6/15 ] recv: S 5
[Send FPGA part 7/15 ] recv: S 6
[Send FPGA part 8/15 ] recv: S 7
[Send FPGA part 9/15 ] recv: S 8
[Send FPGA part 10/15 ] recv: S 9
[Send FPGA part 11/15 ] recv: S 10
[Send FPGA part 12/15 ] recv: S 11
[Send FPGA part 13/15 ] recv: S 12
[Send FPGA part 14/15 ] recv: S 13
[Send FPGA part 15/15 ] recv: S 14
[Send @PHASE_FINE 0x0] recv S 0x1
[Send @TRG_ADD 0x0] recv S 0x1
[Send @trg_holdoff_ext_ADD 0xffff8002] recv S 0x1
[Send @pulse_level_ch1_ADD 0xf19] recv S 0x1
[Send @channel_ch1_ADD 0xffffffa2] recv S 0x3
[Send @volt_gain_ch1_ADD 0x224] recv S 0x3
[Send @zero_off_ch1_ADD 0x21d] recv S 0x3
[Send @CHL_ON_ADD 0x3] recv S 0x1
[Send @channel_ch2_ADD 0xffffff82] recv S 0x3
[Send @volt_gain_ch2_ADD 0x226] recv S 0x3
[Send @zero_off_ch2_ADD 0x215] recv S 0x3
[Send @DM_ADD 0x13ec] recv S 0x1
[Send @SYNCOUTPUT_ADD 0x0] recv S 0x1
[Send @TIMEBASE_ADD 0x190] recv S 0x1
[Send @SAMPLE_ADD 0x0] recv S 0x1
[Send @SLOWMOVE_ADD 0x0] recv S 0x1
[Send @PRE_TRG_ADD 0x9f6] recv S 0x1
[Send @SUF_TRG_ADD 0x9f6] recv S 0x1
[Send @EMPTY_ADD 0x1] recv S 0x4e554c4c
```

florentbr, 2024-08-11:
Have you tried with java11 or java17 ?
What is your distribution and desktop?
There's no issue in the logs.

roopej, 2024-10-07:
I had the same issue with Ubuntu 24.04 LTS. It appears by default I had this Java version:

```~$ java -version
openjdk version "1.8.0_422"
OpenJDK Runtime Environment (build 1.8.0_422-8u422-b05-1~24.04-b05)
OpenJDK 64-Bit Server VM (build 25.422-b05, mixed mode)
```
Instead of that I installed Java 17 by:

```
sudo apt install openjdk-17-jre-headless
sudo apt install openjdk-17-jdk

~$ java -version
openjdk version "17.0.12" 2024-07-16
OpenJDK Runtime Environment (build 17.0.12+7-Ubuntu-1ubuntu224.04)
OpenJDK 64-Bit Server VM (build 17.0.12+7-Ubuntu-1ubuntu224.04, mixed mode, sharing)
```
Tested it for some time and could not make it crash anymore by moving the trigger line or changing time base with +/-. Seems like Java 17 is the way to go.