florentbr/OWON-VDS1022 #103: How to use live plotting

23875658, 2024-10-03:
Hi.

In the vds1022.ipynb, example 3 is titled "Live plotting". I would like to stream the data and plot directly as the OWON application does, but this does not seem like the way to do so. Please assist as I would like to pull voltage and time values directly from the oscilloscope (on both channels) to plot and use this data for real-time analysis.

Is this possible? Thanks for your time.

florentbr, 2024-10-03:
If it no longer work, then it's likely due to changes with the [bokeh](https://docs.bokeh.org/en/latest/)  library. It shouldn't be too hard to fix. Unfortunately I don't plan on spending time on this project.

23875658, 2024-10-03:
Thanks for the quick response, so the live plotting section is the right section to be referring to in vds1022.ipynb for my needs? So the plot should update in real time in a jupyter notebook? I will attempt to make it work.

florentbr, 2024-10-03:
Yes, in the live section the chart should refresh automatically at the given frequency.