florentbr/OWON-VDS1022 #111: File format of Waveform BIN files?

Elmue, 2025-07-03:
Hello

Is there a documentation about the file format of the BIN files that store the Waveform's ?
Or is this an OWON proprietary undocumented format ?
In the latter case, do you know of any project that tried to reverse engineer this file format?

florentbr, 2025-07-06:
> But what can a user expect from a cheap oscillocope that has only 5kB buffer ?
My Rigol has 24 MB sample buffer.

It's a bit harsh given the price difference, especially since it's not meant for data logging.
You get pretty much what an analogue oscilloscope had to offer and more but for a fraction of the cost.
Analogue electronic has not changed since then. I'm sure you can appreciate what a representation over time, even short can bring when creating or debugging a circuit. It is still the cheapest oscilloscope to this date with pulse width triggering. Sure having more memory can be useful, especially when probing digital. But a logic analyzer is in most cases better-suited for the task.

Elmue, 2025-07-06:
You are right, it is like an old analog oscilloscope.
But if the oscilloscope is cheap or not: I was writing this in the context of the quality of the software that OWON gives you.
This is also "cheap". The typical Chinese programming: Quick and Dirty.

> But a logic analyzer is in most cases better-suited for the task.

With a logic analyzer you have other limits.
You cannot see the original signal.
You never know if the A/D conversion worked correctly.
In case of complicated signals that have not been digitized correctly you simply see garbage bytes and don't even notice this.

On the other hand with an oscilloscope with more memory together with Oszi Waveform Analyzer you can do tasks that are completely impossible with ANY commercial logic analyzer, even the most expensive one:

You can capture a bidirectional UART data transfer on one single data line and split it into sent data and received data:
https://netcult.ch/elmue/Oszi-Waveform-Analyzer/#SplitHalfDuplex

[image]

Here you see the result in Oszi Waveform Analyzer. This is the bidirectional serial communication of a pinpad with a chipcard that happens over the same contact on the card and has been split into received and sent data. I never heard of any commercial logic analyzer that could do that.